dani007200964/Shellminator #7: TODO list for V1.2.0

dani007200964, 2022-09-15:
# Todo list for the next release!

- [ ] Override callback functions for the new shortcut keys
- [ ] Option to set the timeout of the internal client
- [ ] Advanced search functions
- [ ] Banner text change option.
- [ ] Path text change option. Also change it to blue color
- [ ] Websocket channel
- [ ] xTerm.js example
- [ ] Terminal buzzer command, with mute option
- [ ] QR-code generation with stronger devices

dani007200964, 2022-09-15:
Page-Up and Page-Down search function prototype added in badc44d995859c7b050912d2cbe9299a4f19e4bb

dani007200964, 2022-09-15:
Crude prototype added for the reverse-i-search function.

https://user-images.githubusercontent.com/26076131/190501257-2e3c3b09-2503-4e33-ae7c-66663c4ecc4e.mp4

Flickering has to be fixed. Also it does not do anything useful with the found command, so it's useless right now.

0xAF, 2022-09-21:
Hi Daniel,
Sorry for the delay.
I was busy with other projects, but eventually got to the hobby again.
Thanks for the great work, you've done so far.
I just tested the dev branch and everything so far seems to be working as charm.
The PgUp/PgDn and the Ctrl-R searches are working nice and good.

Perhaps a small `history` function will be of use for someone, if the buffer gets larger.
I will send you a push request this night I think.

I wanted to say, that I'm not lost. I've been on other stuff, but I'm keeping an eye on your project and will try to contribute as soon as I can.

dani007200964, 2022-09-22:
Hi Stanislav,

> I wanted to say, that I'm not lost. I've been on other stuff, but I'm keeping an eye on your project and will try to contribute as soon as I can.

Of course there are priorities, you have helped me pretty much. Thank you for that! :blush::beers:

> I was busy with other projects, but eventually got to the hobby again.

Me too, I hope your projects are going well.

> Perhaps a small history `function` will be of use for someone, if the buffer gets larger.
I will send you a push request this night I think.

I think it has to be implemented in Commander-API as you said. I'm working on some premade functions for Commander-API. The history function will be one of them.

0xAF, 2022-09-22:
> I think it has to be implemented in Commander-API as you said. I'm working on some premade functions for Commander-API. The history function will be one of them.

Yeah, I was thinking the same.
Perhaps a method to expose the current history from Shellminator would be nice. Then Commander-API or any custom command could use this array to show the history.
And one more thing.
I saw the use of `?` char in the end of the command. It's nice, but It would be nice to have some note of that in the `help` command.
And one more thing, which I find to be some sort of usual in most mini shells.
The `?` as command alone, which is an alias to `help` command.
Most people would just type `?` and expect to get some help.
:)
I'll be back on the hobby track soon.
My other projects are doing well, so I hope soon I will have more time and be able to help you a bit.

dani007200964, 2022-09-27:
Just a small update with the WebSocket version.

https://user-images.githubusercontent.com/26076131/192574049-ef3f5d09-716a-4523-af13-4aa2632067d5.mp4

I have to work a lot on this, but seems usable already.

closed 2023-02-10 by dani007200964