dani007200964/Shellminator #10: Why depend on HardwareSerial?

ondras12345, 2022-10-01:
I wanted to see if I could use your libraries (Shellminator and Commander-API) to create a simple CLI on a STM32F4. However, I use USB CDC as Serial, and the libraries seem to depend on `HardwareSerial`.

**Describe the solution you'd like**
Use Arduino's abstract `Stream` class instead (unless you specifically depend on `HardwareSerial`-specific functionality).

**Describe alternatives you've considered**
I've replaces all occurrences of `HardwareSerial` in both libraries' source code with `Stream`. This brute-force approach seemed to work, but I don't dare use that in production.

dani007200964, 2022-10-01:
It is an excellent idea, I thought about it. I only did this way to support any kind of peripheral to make it work outside Arduino IDE.
Let me work on a prototype for a bit.

ondras12345, 2022-10-02:
Thanks!
I've tested the updated versions of both libraries with my project and they seem to work just fine (after editing my code to adjust for the API changes). I'll play with it some more and sure let you know if I come across a bug.

closed 2023-02-10 by dani007200964