dani007200964/Shellminator #16: Enable formatting

maxbrito500, 2023-04-08:
**Description**
Some consoles like the internal serial on PlatformIO or Arduino 2.x are not able to parse color text compatible with the VT100 terminals. This causes the text to look garbled since the escape characters for colors become printed as text on the screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile the code on either PlatformIO or Arduino IDE
2. Interact with Shellminator using the built-in serial
3. See the garbled text on the "help" menu

**Expected behavior**
There is a global boolean that disables color formatting. This works for the command prompt, but is ignored for internal and external commands. For example the "help" menu and whenever we define color texts inside the functions.

maxbrito500, 2023-04-08:
Detailed examples can be found here: https://github.com/dani007200964/Shellminator/discussions/15#discussioncomment-5558614

dani007200964, 2023-04-08:
I have fixed the help format problem with 2489648749cb579ae3481b6a4d9a3158f512c89d, but the static version of __setTerminalCharacterColor__ is tricky. Because it is static, it doesn't have an actual object. Because of this, it does not have a __enableFormatting__ flag. This way, the only solution right now for this is to track the status of that channel manually in your code.

dani007200964, 2023-04-09:
A small update. It probably breaks backward compatibility 😢 

I modified Shellminator and Commander a bit to make it easy to refer to the shell object, that called the execution-functions.
It is in the dev branch currently and works very simply. There is a third argument in Commander's API functions called the parent. Shellminator passes a pointer to it from the shell object, which is called this command. I can demonstrate it easily with an example.

```c
void format_func(char *args, Stream *response, void* parent )
{

  // Pointer to shell object if possible.
  Shellminator* shell;

  // argument value
  int value;

  // This variable will hold the result of the
  // argument parser.
  int argResult;

  if( parent ){

    shell = Shellminator::castVoidToShellminator( parent );

    argResult = sscanf( args, "%d", &value );

    if( argResult != 1 ){

      response -> print( "Argument error!\r\n" );
      return;

    }

    shell -> enableFormatting = !!value;
    return;

  }

  response -> print("Shell object not available!\r\n");

}
```

This way, you can access the terminal object from the command function. My only concern is, that this will break backward compatibility, but it was a feature, that I wanted to add for a long time. It makes it very easy to manipulate a specific shell.

What do you think?

It is currently in development and probably has a ton of bugs. Also, I worked on the formatting problem both with Shellminator and Commander-API. It seems to work fine with the simulator. It is tricky to get rid of the garbage characters because a lot of functionality relies on them. I think it is now good enough for the Arduino IDE. Please check it if you have some spare time 😄

dani007200964, 2025-03-08:
Solved in V3

closed 2025-03-08 by dani007200964