dani007200964/Shellminator #27: "Compilation error: no match for 'operator=' in Shellminator-BLE-Stream.cpp"

xar-pb, 2025-03-13:
**Describe the bug**
In newest version of C++ libraries, Android IDE and Shellminator library the basic example does not compile.

**To Reproduce**
Steps to reproduce the behavior:
1. The used board is: 'ESP32 Dev Module'
2. The used microcontroller is: 'ESP32-WROOM-32 
3. The used environment and it's version is: 'Arduino IDE 2.3.4'
4. The used OS on your computer is: 'Windows 11'

**Expected behavior**
Basic example is not compiled correctly. Error displayed:

d:\Data\Dokumenty\Arduino\libraries\Shellminator\src\Shellminator-BLE-Stream.cpp: In member function 'virtual void ShellminatorBleStream::bleRxCallback::onWrite(BLECharacteristic*)':
d:\Data\Dokumenty\Arduino\libraries\Shellminator\src\Shellminator-BLE-Stream.cpp:170:43: error: no match for 'operator=' (operand types are 'std::string' {aka 'std::__cxx11::basic_string<char>'} and 'String')
  170 |             dataStr = charPtr -> getValue();
      |                                           ^
In file included from C:/Users/petrb/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/string:54,
                 from C:\Users\petrb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\BLE\src/BLEDevice.h:18,
                 from d:\Data\Dokumenty\Arduino\libraries\Shellminator\src\Shellminator-BLE-Stream.hpp:50,
                 from d:\Data\Dokumenty\Arduino\libraries\Shellminator\src\Shellminator-BLE-Stream.cpp:34:

**Working solution**
file:
Shellminator-BLE-Stream.cpp 
L170: original line (error): 
dataStr = charPtr -> getValue();

L170:  replace original line with new line (fix): 
dataStr = std::string(charPtr->getValue().c_str());

dani007200964, 2025-03-13:
Thank you!

Interesting, on my side it compiles with my setup:

[image]

I see that, you have a newer version of the toolchain, I update mine and I will check it.

xar-pb, 2025-03-13:
Thank you, sending screen from my Arduiono IDE with original Shellminator-BLE-Stream.cpp L170: dataStr = charPtr -> getValue(); 
,after fix to L170: dataStr = std::string(charPtr->getValue().c_str());
example compiles OK.

[image]

dani007200964, 2025-03-13:
Yep, I'm already tried with ESP32 Core V3.1.1 and the error come up on my side too. It seems like that the BLE API has been refactored in ESP32 Core since last time when I updated it( maybe a year ago, who knows 🤔 ).

Thank you, it seems like with your fix it is working, at least it's compiling now. I need to check the BLE functionality as well.

xar-pb, 2025-03-13:
Great, thank you very much. I really admire the work you did on Shellminator.

dani007200964, 2025-03-13:
@xar-pb Thank you, it seems like it is working with your fix. I tried with the BLE terminal as well and no issues on my side:

[image]

I will create a bugfix release for this, but it takes at least a night to get it synced to the servers. Until then, please use your version, or install it manually. I need maximum an hour to do this.

closed 2025-03-13 by dani007200964